ThomasLengeling/KinectPV2 #12: Error in PointCloudOGL

kokashking, 2015-02-01:
Hi first of all thanx for a great library.

My system: 
Win 8.1 64x, Processing 2.2.1, DirectX 11, i7-4710HQ, NVIDIA GTX 980m

after a couple of minutes PointCloudOGL example crasches with:

*java.lang.RuntimeException: java.lang.IndexOutOfBoundsException: Required 1 remaining elements in buffer, only had 651264
	at com.jogamp.common.util.awt.AWTEDTExecutor.invoke(AWTEDTExecutor.java:58)
	at jogamp.opengl.awt.AWTThreadingPlugin.invokeOnOpenGLThread(AWTThreadingPlugin.java:103)
	at jogamp.opengl.ThreadingImpl.invokeOnOpenGLThread(ThreadingImpl.java:206)
	at javax.media.opengl.Threading.invokeOnOpenGLThread(Threading.java:172)
	at javax.media.opengl.Threading.invoke(Threading.java:191)
	at javax.media.opengl.awt.GLCanvas.display(GLCanvas.java:541)
	at processing.opengl.PJOGL.requestDraw(PJOGL.java:688)
	at processing.opengl.PGraphicsOpenGL.requestDraw(PGraphicsOpenGL.java:1651)
	at processing.core.PApplet.run(PApplet.java:2256)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.IndexOutOfBoundsException: Required 1 remaining elements in buffer, only had 651264
	at com.jogamp.common.nio.Buffers.rangeCheck(Buffers.java:1033)
	at jogamp.opengl.gl4.GL4bcImpl.glVertexPointer(GL4bcImpl.java:35737)
	at PointCloudOGL.draw(PointCloudOGL.java:91)*

ThomasLengeling, 2015-02-03:
Yeah I notice that once. I just need to get the time to update it, something is wrong when transferring the data from C++ to Java.

ThomasLengeling, 2015-07-09:
Try out the new version,

closed 2015-07-09 by ThomasLengeling