ThomasLengeling/KinectPV2 #13: banding in depth images

shiffman, 2015-03-25:
I'm seeing a lot of strange bands in my depth image similar to this image on your site.

[image]

This doesn't seem right to me as it should be a continuous grayscale with 0 as far away and 255 as close.   Is this part of something with the new kinect and I'm mistaken?

ThomasLengeling, 2015-03-25:
For some reason on the Kinect SDK the grayscale banding is the default method of obtaining the depth Image. But I can provide a similar method for obtaining the continuous grayscale from 0-255.

[image]

shiffman, 2015-03-25:
Interesting, I guess it cycles through the brightness values to show depth detail (since there are more than 255 measurements).  I think this is less useful for the end user in the "simple Processing sense" so adding a method for continuous grayscale would have value I think.

I suppose the "correct" way to threshold for depth would be to use the raw depth data then?

jubby, 2015-03-26:
Hey, I've fixed the depth problem.  I just need access to upload the fix.  Thomas, can you give me "Jubby" permissions.  I've created a separate branch for the fix.

ThomasLengeling, 2015-03-26:
Hey jubby, great that you found that, I was planning just to a new method.
Can you make a pull request with the modified code?
Thomas

jubby, 2015-03-26:
Yep, that’s what I did, created a new method for just the raw 16-bit data stream.  I’ll do the pull request when I get home tonight.

Thanks,
Kirk

jubby, 2015-03-30:
Sorry I didn’t get this in yet.  Something came up over the weekend.  I’ll get it in ASAP.

Thanks,
Kirk

ThomasLengeling, 2015-07-09:
Fixed with the new version, also the raw data

closed 2015-07-09 by ThomasLengeling