ThomasLengeling/KinectPV2 #15: unsupported Java version issue?

neokio, 2015-04-26:
Hey, I'm getting the following error when running any of the sketches:

   UnsupportedClassVersionError: A library is using code compiled with an unsupported version of Java.

    This version of Processing only supports libraries and JAR files compiled
    for Java 1.6 or earlier. A library used by this sketch was compiled for
    Java 1.7 or later, and needs to be recompiled to be compatible with Java 1.6.

I'm running Processing 2.2.1 on Windows 8.1, and the latest Kinect v2 SDK. I tried installing Java 1.8, 1.7 and 1.6 before re-reading the error and realizing that wouldn't help.

We're excited to see what you've done! Any advice on getting this running?

ThomasLengeling, 2015-04-27:
mm, strange.

try https://github.com/ThomasLengeling/KinectPV2/tree/002cad2d5f531087852662177d852b3c75d758fb

neokio, 2015-04-27:
Hi,
Yep that tree works.
Thanks :)

sidhant-goyal, 2015-06-01:
Hey, i get this error when i run most of the processing examples of your library, any suggestion?

sidhant-goyal, 2015-06-01:
64 windows 8
Loading KinectV2
Creating Kinect object ...
ENABLE DEPTH FRAME
ENABLE INFRARED FRAME
ENABLE LONG EXPOSURE INFRARED
Done init Kinect v2
Version: 0.7.2
EXIT
Clossing kinect V2
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ffdc95b4110, pid=18944, tid=14212
#
# JRE version: Java(TM) SE Runtime Environment (7.0_40-b43) (build 1.7.0_40-b43)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (24.0-b56 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [KinectPV2.dll+0x4110]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# D:\Design\Assignments\Sem 6\TUI\India HCI\processing-2.2.1-windows64\processing-2.2.1\hs_err_pid18944.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.

ThomasLengeling, 2015-06-01:
do you have the hs_err_pid18944.log file?,

The current release  should work, do you have the kinect SDK installed ?

sidhant-goyal, 2015-06-02:
Yes i have the file and the kinect SDK2 installed, but i cannot find the prime sensor installed in my device manager. Can you please share the link and also go through the error file?

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ffdc95b4110, pid=18944, tid=14212
#
# JRE version: Java(TM) SE Runtime Environment (7.0_40-b43) (build 1.7.0_40-b43)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (24.0-b56 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [KinectPV2.dll+0x4110]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x000000000cccf800):  JavaThread "Animation Thread" [_thread_in_native, id=14212, stack(0x000000000f200000,0x000000000f300000)]

siginfo: ExceptionCode=0xc0000005, reading address 0xffffffffffffffff

Registers:
RAX=0x0000000000000001, RBX=0x0000000021e33040, RCX=0xffffffffffffffff, RDX=0x0000000000000000
RSP=0x000000000f2feb00, RBP=0x0000000000000006, RSI=0x0000000000000000, RDI=0x000000001034a348
R8 =0x000000000f2fe7c8, R9 =0x0000000000008000, R10=0x0000000000000000, R11=0x000000000f2fea00
R12=0x0000000000000000, R13=0x000000077d1f8ae0, R14=0x0000000000000000, R15=0x000000000cccf800
RIP=0x00007ffdc95b4110, EFLAGS=0x0000000000010286

Top of Stack: (sp=0x000000000f2feb00)
0x000000000f2feb00:   000000001034a318 000000001033ff48
0x000000000f2feb10:   000000001034a318 0000000000000000
0x000000000f2feb20:   000000000f2febf0 00007ffdc95b1fba
0x000000000f2feb30:   000000000f2febf0 000000000f2febd0
0x000000000f2feb40:   000000000cccf9e8 000000000cc26fc0
0x000000000f2feb50:   0000000002236198 0000000002249949
0x000000000f2feb60:   000000077d1f8ae0 0000000000000002
0x000000000f2feb70:   000000000f2fec90 000000000f2fec48
0x000000000f2feb80:   000000000cccf800 000000000f2fec48
0x000000000f2feb90:   000000000f2feb90 000000077d1f8ae0
0x000000000f2feba0:   000000000f2febf0 000000077d1fad70
0x000000000f2febb0:   0000000000000000 000000077d1f8ae0
0x000000000f2febc0:   0000000000000000 000000000f2febf0
0x000000000f2febd0:   000000000f2fec38 0000000002236198
0x000000000f2febe0:   0000000000000000 0000000002243a58
0x000000000f2febf0:   0000000785a81060 000000000f2febf8 

Instructions: (pc=0x00007ffdc95b4110)
0x00007ffdc95b40f0:   89 77 d0 48 8b 0f 48 85 c9 74 09 48 8b 01 ff 50
0x00007ffdc95b4100:   10 4c 89 37 48 8b 8f 90 00 00 00 48 85 c9 74 0d
0x00007ffdc95b4110:   48 8b 01 ff 50 10 4c 89 b7 90 00 00 00 48 8b 4f
0x00007ffdc95b4120:   30 48 85 c9 74 0a 48 8b 01 ff 50 10 4c 89 77 30 

Register to memory mapping:

RAX=0x0000000000000001 is an unknown value
RBX=0x0000000021e33040 is an unknown value
RCX=0xffffffffffffffff is an unknown value
RDX=0x0000000000000000 is an unknown value
RSP=0x000000000f2feb00 is pointing into the stack for thread: 0x000000000cccf800
RBP=0x0000000000000006 is an unknown value
RSI=0x0000000000000000 is an unknown value
RDI=0x000000001034a348 is an unknown value
R8 =0x000000000f2fe7c8 is pointing into the stack for thread: 0x000000000cccf800
R9 =0x0000000000008000 is an unknown value
R10=0x0000000000000000 is an unknown value
[... 320 lines cut ...]
VM Arguments:
jvm_args: -agentlib:jdwp=transport=dt_socket,address=8960,server=y,suspend=y -Djava.library.path=C:\Users\SidGoyal\Documents\Processing\libraries\KinectPV2\examples\DepthTest\code;C:\Users\SidGoyal\Documents\Processing\libraries\KinectPV2\library;D:\Design\Assignments\Sem 6\TUI\India HCI\processing-2.2.1-windows64\processing-2.2.1\java\bin;C:\windows\system32;C:\windows;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client;C:\Program Files\Intel\iCLS Client;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\WiFi\bin;C:\Program Files\Common Files\Intel\WirelessCommon;C:\Program Files\Condusiv Technologies\ExpressCache;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Python27;C:\Python27\Lib\site-packages;C:\Program Files\nodejs;C:\Users\SidGoyal\AppData\Roaming\npm;C:\Program Files (x86)\nodejs;C:\Program Files\nodejs;C:\Program Files (x86)\Git\cmd;C:\Program Files (x86)\Skype\Phone;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit;C:\Program Files\Microsoft SQL Server\110\Tools\Binn;C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0;C:\Program Files\Microsoft SQL Server\120\Tools\Binn;C:\Users\SidGoyal\AppData\Local\Android\android-sdk\platform-tools;C:\Users\SidGoyal\AppData\Local\Android\android-sdk\tools;C:\All\apache-ant-1.9.4-bin\apache-ant-1.9.4\bin;C:\Users\SidGoyal\AppData\Roaming\npm;C:\Program Files\Java\jdk1.8.0_40\bin;D:\Design\Assignments\Sem 6\TUI\India HCI\processing-2.2.1-windows64\processing-2.2.1\java\bin;. -ea 
[... cut ...]
OS: Windows 8 , 64 bit Build 9200 

CPU:total 8 (4 cores per cpu, 2 threads per core) family 6 model 60 stepping 3, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, avx2, aes, erms, ht, tsc, tscinvbit

Memory: 4k page, physical 8298776k(2572056k free), swap 11588628k(3904668k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (24.0-b56) for windows-amd64 JRE (1.7.0_40-b43), built on Aug 26 2013 22:38:32 by "java_re" with unknown MS VC++:1600

time: Mon Jun 01 19:21:02 2015
elapsed time: 37 seconds

ThomasLengeling, 2015-06-02:
The links are un the readme file, chevk if you hace all the requirements.  And try the kinect SDK examples, if they work kinectpv2 should work without a problem

gwyllo, 2015-06-12:
Hi Thomas, 

Thanks for such an excellent library - its been really useful to us as a robotic vision tool. We are working with the most recent version from github.

We have started playing around with the getPointCloudColor and getColorChanelBuffer functions in the library to create point cloud masks based on RGB colours coming from the kinect. Working in eclipse, these functions tend to cause java to crash ~50% of the time due to a memory access error that seems similar to the problem faced by the user above. Eclipse prints this to the console:

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ffb8a02498d, pid=41428, tid=42972
#
# JRE version: Java(TM) SE Runtime Environment (7.0_71-b14) (build 1.7.0_71-b14)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (24.71-b01 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [KinectPV2.dll+0x498d]

The library seems to run completely smoothly when working with black and white depth point clouds, or 2d colour - but the functions that combine these throw errors. I was hoping you could shed some insight?

Thanks,
Gwyll

closed 2015-07-09 by ThomasLengeling