ThomasLengeling/KinectPV2 #21: your point cloud Unable to run continuously

zuojiadi, 2015-07-14:
Thank you for your library.
I have a trouble.
It runs after period of time
Tell me the class gl2 doesn't exist ..
It is really running for a period of time.About three or four minutes

ThomasLengeling, 2015-07-14:
have you tried the latest version?, 0.7.3

zuojiadi, 2015-07-14:
I tried just now,
Do not work correctly.

java.lang.RuntimeException: javax.media.opengl.GLException: Caught UnsupportedClassVersionError: KinectPV2/KinectPV2 : Unsupported major.minor version 52.0 on thread AWT-EventQueue-0
	at com.jogamp.common.util.awt.AWTEDTExecutor.invoke(AWTEDTExecutor.java:58)
	at jogamp.opengl.awt.AWTThreadingPlugin.invokeOnOpenGLThread(AWTThreadingPlugin.java:103)
	at jogamp.opengl.ThreadingImpl.invokeOnOpenGLThread(ThreadingImpl.java:201)
	at javax.media.opengl.Threading.invokeOnOpenGLThread(Threading.java:202)
	at javax.media.opengl.Threading.invoke(Threading.java:221)
	at javax.media.opengl.awt.GLCanvas.display(GLCanvas.java:525)
	at processing.opengl.PJOGL.requestDraw(PJOGL.java:698)
	at processing.opengl.PGraphicsOpenGL.requestDraw(PGraphicsOpenGL.java:1656)
	at processing.core.PApplet.run(PApplet.java:2260)
	at java.lang.Thread.run(Unknown Source)
Caused by: javax.media.opengl.GLException: Caught UnsupportedClassVersionError: KinectPV2/KinectPV2 : Unsupported major.minor version 52.0 on thread AWT-EventQueue-0
	at javax.media.opengl.GLException.newGLException(GLException.java:75)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1311)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1131)
	at javax.media.opengl.awt.GLCanvas$11.run(GLCanvas.java:1394)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$200(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: java.lang.UnsupportedClassVersionError: KinectPV2/KinectPV2 : Unsupported major.minor version 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(Unknown Source)
	at java.security.SecureClassLoader.defineClass(Unknown Source)
	at java.net.URLClassLoader.defineClass(Unknown Source)
	at java.net.URLClassLoader.access$100(Unknown Source)
	at java.net.URLClassLoader$1.run(Unknown Source)
	at java.net.URLClassLoader$1.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at PointCloudOGL.setup(PointCloudOGL.java:52)
	at processing.core.PApplet.handleDraw(PApplet.java:2365)
	at processing.opengl.PJOGL$PGLListener.display(PJOGL.java:873)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:690)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:672)
	at javax.media.opengl.awt.GLCanvas$10.run(GLCanvas.java:1383)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1277)
	... 16 more

My technology is very poor.
It means that there are some things I didn't configured right

ThomasLengeling, 2015-07-14:
you need to use processing 3.0

zuojiadi, 2015-07-14:
Thank you very much.
I'm sorry, wasted your time.I always thought processing 3.0a5 is 3.0.

 it must run in 3.0 a10

closed 2015-07-19 by ThomasLengeling