ThomasLengeling/KinectPV2 #22: KinectPV2 java issues

Adiou, 2015-08-03:
Hi, 
when I try the example, i get "This version of Processing only supports libraries and JAR files compiled for Java 1.6 or earlier. A library used by this sketch was compiled for Java 1.7 or later, 
and needs to be recompiled to be compatible with Java 1.6."

I saw this already happend to somebody else. You provided him an other download link, it work for him, he doesn't for me...

I'm on windows 8.1, with the latest kinect sdk and processing 2.2.1.

Thanks for your help! 

Adrien

ThomasLengeling, 2015-08-03:
hey!, try using the library with the latest processing 3.11a 

thomas

Adiou, 2015-08-03:
Hey, thanks for your fast answer!
I tried, but I get a" illegalStateException : fullScreen() cannot be used here...."
I'm checking processing reference right now...
Adrien

Adiou, 2015-08-03:
I change the size, but now I get error because I'm using 32bits version of processing 3...Downloading the 64bits versions...

Adiou, 2015-08-03:
Tried it with 64bits version, still "illegalStateException : fullScreen() cannot be used here" pb, change the size, sketch running but I just get a black screen...

Adiou, 2015-08-03:
It's working with processing 2.2.1 and the version 0.7.1...

ThomasLengeling, 2015-08-03:
yeah is only for windows 8 64 bits, and the newest version of processing.

Sorry for the minor bugs of the examples, the new size() only works with static values, so change size(512*3, 424) to size(1536, 424).

I'll to try to make those fixes today.

Thomas

Adiou, 2015-08-03:
Thanks for all Thomas!
Adrien

Adiou, 2015-08-03:
Just a last thing : with processing 3 and the version 0.7.3 of the library, when I change the size, the sketch is running but I have nothing but a black screen with the frameRate...

ThomasLengeling, 2015-08-03:
amm strange. any console output?

Adiou, 2015-08-03:
Actually BodyTrack_User, DepthTest, HDColor,  HDFaceVertex, MapDepthToColor, Mask_findUsers, MaskTest, FindContours, LiveCamTest, PointCloudColor, PointCloudDepth, PointCloudOGL, RecordPointCloud, SimpleFaceTracking, Skeleton3d, skeletonColor, SkeletonMaskDepth and TestImage are actually working.

But for CoordinateMapperRGBDepth i get "the function "setCoordbkglmg(int[])" does not exist" (sketch is not running) and for HDFaceVertex the sketch is running but I just get a black screen, no console output.

ThomasLengeling, 2015-08-03:
hey the CoordinateMapperRGBDepth  is not working use 0.7.2, the HDFaceVertex should be working, takes some time in finding the user, or you should be close to the kinect/

Adiou, 2015-08-03:
Yeap, you're right ;-)

ThomasLengeling, 2015-08-04:
updated readme

closed 2015-08-04 by ThomasLengeling