ThomasLengeling/KinectPV2 #28: Head orientation?

Adiou, 2015-08-21:
Hi Thomas,
Is it already possible to get head orientation?
Thx a lot for your good job!
A.

ThomasLengeling, 2015-08-21:
Yeah, check the simple face detection example

Adiou, 2015-08-22:
Thx for your answer, but I don't see it...

Adiou, 2015-08-22:
And Is it normal that even if I download and install version 0.7.5, when I run sketches consol says Version : 0.7.4?
Cheers, 
A.

ThomasLengeling, 2015-08-22:
ahh great that you point that out, forgot to change the lib version in the lib.

Yeah you are right, its actually implemented but the getters for yaw, pitch and roll a missing, sorry about that.

ThomasLengeling, 2015-08-22:
wait for an update soon

Adiou, 2015-08-23:
Ok, thanks!

Adiou, 2015-08-31:
Hi Thomas, 
Do you know if you'll have time to share a new update with face orientation this week?
Thanks a lot, 
Take care, 
A.

ThomasLengeling, 2015-09-02:
Hey Adiou, I just update the lib, try the current github version, check out the example simplefacetracking.

```
      //head orientation
      float pitch = faceD.getPitch();
      float yaw   = faceD.getYaw();
      float roll  = faceD.getRoll();
```

the Kinect SDK throws a quarternion for the rotation, so the pitch, yaw and roll are in degrees.

```
		double x = pQuaternion->x;
		double y = pQuaternion->y;
		double z = pQuaternion->z;
		double w = pQuaternion->w;

		// convert face rotation quaternion to Euler angles in degrees		
		dPitch = atan2(2 * (y * z + w * x), w * w - x * x - y * y + z * z) / M_PI * 180.0;
		dYaw = asin(2 * (w * y - x * z)) / M_PI * 180.0;
		dRoll = atan2(2 * (x * y + w * z), w * w + x * x - y * y - z * z) / M_PI * 180.0;
```

I haven't tested it out, so let know it works for you

thomas

Adiou, 2015-09-02:
Hi, thanks a lot Thomas, 
I'll try to test this today or at least this week, I'll let you know!
Thanks again, 
Adrien

Adiou, 2015-09-03:
Hi Thomas, 

I test a bit head orientation and  it's working very well, thanks!

BUT, I have a pb : the frameRate is slowing down contiunously. I thougth it was my sketch but actually it appear in some exemples aswell (with skeleton or simpleFaceTracking for exemple) : it start around 60, and after a couple of minutes, it is under 20. Sometimes even if the kinect is not seeing me...
I tested this with BodyTrack_users exemple and obviously it's not happening (frameRate start around 40/30 but it stay at this level...).
I'll test this a bit more and will let you know, if you get any question or idea, let me know...

ThomasLengeling, 2015-09-03:
I have notice that in a couple of examples, I need to track it down. On the weekend I'll check the skeleton tracking code as well as the face tracking, they use similar code structure.

thank you

Adiou, 2015-09-17:
Hi, 
Any clues about the slowing down problems?
Cheers, 
A.

ThomasLengeling, 2015-09-17:
hey, sorry haven't gotten time to check it out, ill try to do it the next days.

Marce1o, 2016-06-26:
the problem with the low frame rate was directX12, I think