ThomasLengeling/KinectPV2 #30: getPointCloudColorPos & getPointCloudDepthPos returns zeros

Skofo1, 2015-09-02:
//get the points in 3d space
  FloatBuffer pointCloudBuffer = kinect.getPointCloudDepthPos();

  // obtain XYZ the values of the point cloud
  for(int i = 0; i < kinect.WIDTHDepth * kinect.HEIGHTDepth; i++){
     float x = pointCloudBuffer.get(i*3 + 0);
     float y = pointCloudBuffer.get(i*3 + 1);
     float z = pointCloudBuffer.get(i*3 + 2);
  }

returns zero for every x,y,z. Are you getting this too?

ThomasLengeling, 2015-09-02:
ahh ok so it works?, the recordpointcloud example uses similar to that

Skofo1, 2015-09-02:
yep, I was able to create a point cloud without the use of shaders with it, so this actually gives you the correct values! :)

ThomasLengeling, 2015-09-02:
great!

closed 2015-09-02 by ThomasLengeling

Skofo1, 2015-09-02:
Hey Thomas,

can you point me in a direction on how to add color to the point cloud without the use of shaders? I’ve had a look at your getMapDepthToColor() method, because I thought it would give me an array with color values for every „depthpixel“, yet it’s twice as long as a deptharray. What exactly does this method return? Do you have any other advice for mapping the colorimage onto the point cloud?

All the best
Moritz