ThomasLengeling/KinectPV2 #35: Banding in long exposure IR image

JakubValtar, 2015-11-11:
Hi Thomas, firstly thanks for your work on this, your library is making my life much easier.

The issue here is that I see banding in long exposure IR image similar to what was reported in #13.

See bottom right. Close distances are getting black again.
[image]

I believe it's because [here](https://github.com/ThomasLengeling/KinectPV2/blob/master/Build_libs/KinectPV2_vc2012/KinectLib_V2.0/KinectPV2.cpp#L1021) it's shifted by 6 instead of 8, so two highest bits get discarded and brightest values of more than 0xC000 appear to start from zero again. However I don't have Visual Studio on my machine right now so I can't test it and send you a pull request.

ThomasLengeling, 2015-11-11:
you are right @JakubValtar!, thank you I'll change  it!!

JakubValtar, 2015-11-11:
Nice, glad I could help :)

closed 2015-11-24 by ThomasLengeling