ThomasLengeling/KinectPV2 #37: How can I close the kinect v2？？？

rongloveyan, 2015-11-24:


ThomasLengeling, 2015-11-24:
try using closeDevice() function.
Let me know how it works

rongloveyan, 2015-11-25:
64 windows 8.1
Loading KinectV2
Creating Kinect object ...
Setting Color Frame
Setting Skeleton
Done Kinect V2 
Version: 0.7.6
Clossing kinect V2
Creating Kinect object ...
Setting Color Frame
Setting Skeleton
Done Kinect V2 
Version: 0.7.6
Clossing kinect V2
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ffb90c2554f, pid=4072, tid=6172
#
# JRE version: Java(TM) SE Runtime Environment (8.0_51-b16) (build 1.8.0_51-b16)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.51-b03 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  0x00007ffb90c2554f
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\Rong\Downloads\processing-3.0.1\hs_err_pid4072.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Could not run the sketch (Target VM failed to initialize).
For more information, read revisions.txt and Help → Troubleshooting.

rongloveyan, 2015-11-25:
I tried to use dispose() function,but it had same error.

rongloveyan, 2015-11-25:
I tried to use dispose() function,but it had the same error.