ThomasLengeling/KinectPV2 #38: Export XYZ + Infrared

MartinPetitMarteau, 2016-01-07:
Dear Thomas,
first of all thanks a lot for the work you put into the KinectPV2 project!

What I am searching for is a possibility to map the infrared values to the depth image and to export a file which contains the XYZ coordinates of the points as well as the corresponding grey value of the IR channel for each point.
The RGB values are accessible via 'getColorChannelBuffer()' (as far as I understood), but is there also a solution for tapping the infrared channel in the same way?

Thanks in advance for your answer!

abenzer, 2016-01-08:
@MartinPetitMarteau this project might be helpful https://github.com/najeebt/atrium

ThomasLengeling, 2016-01-08:
ohh see, there are couple of missing mapping functions

MartinPetitMarteau, 2016-01-08:
@abenzer I am looking explicitly for a possibility to extract also the infrared greyscale value for each depth pixel / XYZ coordinate, which does not seem to be included in atrium, but thanks for the hint!

MartinPetitMarteau, 2016-07-11:
Hi again!

Sorry for being stubborn, but would it be a large effort for you to provide a function to access the infrared image so that the following example would be possible?

`FloatBuffer InfraredBuffer = kinect.getInfraredImageBuffer();`

bwjames, 2017-04-14:
@MartinPetitMarteau were you able to make any progress on this? I'm trying to extract the XYZ coordinate data of the points as the Kinect scans, and your question seemed similar. Thanks!