ThomasLengeling/KinectPV2 #48: Issues with vertexAttribPointer() - none of the pointcloud examples are working

paraclete-pizza, 2016-04-02:
Whenever I try to run any of the point cloud examples in the library, I get the following error:

`The function "vertexAttribPointer()" expects parameters like: "vertexAttribPointer(int, int, int, boolean, int, int)"`

Related to these lines of code:
```
  pgl.vertexAttribPointer(vertLoc, 3, PGL.FLOAT, false, 0, pointCloudBuffer);
  pgl.vertexAttribPointer(colorLoc, 3, PGL.FLOAT, false, 0, colorBuffer);
```

I've tried looking through the KinectPV2 Reference, but cannot find the functions involved, and am at a bit of a loss as to what to do.

paraclete-pizza, 2016-04-02:
The issue seems to be with pointCloudBuffer and colorBuffer, neither of which are 'int' values.

I tried appending '.arrayOffset()' to the end of each, which should return an int, and (I thought) provide the 'offset' value vertexAttribPointer is looking for. However, while this allowed the program to compile, I got a runtime error:

`unsuportedOperationException
`

ThomasLengeling, 2016-04-25:
are you still having problems with the openGL examples?

paraclete-pizza, 2016-04-25:
Since I updated from 0.7.5 to 0.7.7, the OpenGL point cloud examples both work fine!

closed 2016-04-25 by paraclete-pizza