ThomasLengeling/KinectPV2 #49: getPointCloudDepthPos returns null, crashes program

paraclete-pizza, 2016-04-23:
I'm trying to access the X, Y, Z coordinates of the point cloud without shaders, so I can do my own things with the data. In the past, I was able to do this with OpenNi and the original Kinect, but I am having a lot of trouble doing the same with Kinect v2 and KinectPV2. All of the included examples rely on opaque OpenGL shaders, when all I really want is some X, Y, Z values to play with.

I tried uncommenting the code in the pointcloudOGL example:

```
  // obtain XYZ the values of the point cloud
  // for(int i = 0; i < kinect.WIDTHDepth * kinect.HEIGHTDepth; i++){
  //    float x = pointCloudBuffer.get(i*3 + 0);
  //    float y = pointCloudBuffer.get(i*3 + 1);
  //    float z = pointCloudBuffer.get(i*3 + 2);
  // }
```
However, doing so, and then trying to draw points with, say, point (x, y, z), freezes the program, as does simply trying to print the values to the console (which yields all 0s for x, y, and z). When I try to use .array() to extract the array, it likewise crashes, and if I use hasArray() on the buffer, it returns false.

I've tried both enablePointCloud and enableColorPointCloud in my setup function, but neither seems to work. Am I doing something wrong, or is this an issue with the library?

ThomasLengeling, 2016-04-25:
I added a quick update to that example where you can see the result of the points in 3D, just un-comment that part.
The floatBuffer is not associated with an array that is way when you try ```hasArray()``` it returns 0, and probably there is so many print calls that the app crashes.

Hope this works

paraclete-pizza, 2016-04-25:
I managed to get it to work! With your code, I was still getting crashes, but from my graphics hardware getting overwhelmed.

I narrowed the range of the depth cloud from a 4m range to a .5m range, and it no longer crashed, but just seemed to show a single large circle.

This turned out to be one point of the cloud, zoomed in very, very far, and after playing with the zoom, scale, and rotation, managed to calibrate it to get a point cloud I could play with. (I also upped the interval of the for loop to i+=10, to help performance).

[image]

closed 2016-04-25 by paraclete-pizza

ThomasLengeling, 2016-04-25:
great man!