ThomasLengeling/KinectPV2 #59: Is there any complete documentation for your library ?

antoine1000, 2017-02-20:
Hi,

First, thanks for your wonderful library. 

 I was asking myself if you made a complete documentation of the library. I mean, your [webpage ](http://codigogenerativo.com/code/kinectpv2-k4w2-processing-library/) is really helping, but a complete documentation with references about how functions works, which paramaters they can take, in which context we can use it...etc, would be wonderful !

Thank you !

antoine1000, 2017-02-20:
Finding it on the reference folder...

antoine1000, 2017-02-21:
Sorry for the comments, but it appears that only FaceData is well referenced, the rest of the library is not...

ecargnfx, 2017-03-04:
Where is the documentation you found? Which files in reference folder?

antoine1000, 2017-03-04:
On the KinectPV2 Library, you have a reference folder : 
<br>
[image]

In there you have multiples html pages, with some documentation, but most of it is very incomplete : 

[image]

If every class was documented it will be incredible ! 👍

ecargnfx, 2017-03-05:
Yea, I agree!