ThomasLengeling/KinectPV2 #60: example "SkeletonColor" why joints[jointType].getZ() always zero

Corllll, 2017-03-20:
i wanna use the depth value.
void drawJoint(KJoint[] joints, int jointType) {
  pushMatrix();
  translate(joints[jointType].getX(), joints[jointType].getY(), joints[jointType].getZ());
  ellipse(0, 0, 25, 25);
  popMatrix();
}
the joints[jointType].getZ() is work?
someone can help me?

mrOzeMus, 2018-03-11:
Hello.

I had the same problem on the exemple SkeletonMaskDepth. By watching the code and the example skeleton3d, (the example doesn't work on my pc but the code make sense) it seems that for getting z value of the skeleton the good way is to use  kinect.getSkeleton3d().

So try in setup: kinect.enableSkeleton3DMap(true);
and replace or make a new arraylist <KSkeleton> skeletonArray = kinect.getSkeleton3d();

and now you should have access to Z values (need to scale the value). It worked for me. Hope it helps.

YANG-Mingcong, 2019-04-21:
Hello.

I have the same problem with you.
I make a new arraylist for that.

code blow
```processing
  ArrayList<KSkeleton> skeletonArray =  kinect.getSkeletonColorMap();
  ArrayList<KSkeleton> skeletonArrayZ =  kinect.getSkeleton3d();            //this is new for getZ()

  //individual JOINTS
  for (int i = 0; i < skeletonArray.size(); i++) {
    KSkeleton skeleton = (KSkeleton) skeletonArray.get(i);
    KSkeleton skeletonZ = (KSkeleton) skeletonArrayZ.get(i);                      //this is new for getZ()
    if (skeleton.isTracked()) {
      text("skeleton is tracked", 50, 90);
      KJoint[] joints = skeleton.getJoints();
      KJoint[] jointsZ = skeletonZ.getJoints();                                                 //this is new for getZ()
```
after that, you can use this code to check if you get the right Z

`println(jointsZ[KinectPV2.JointType_HandLeft].getZ());`

I just have try and the Z seems in unit meter.

ZvivZ, 2022-06-06:
Hi, I am using this method to get the Z depth of a joint, and it works most of the time, except my program crashes randomly with this error when using getSkeleton3d().  I tried getting depth using rawDepthData, but it's too slow.  I'd really like to use the depth of certain joints, but getting them is proving very strange and difficult.  Any help would be appreciated.

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ffb34de5549, pid=280, tid=0x0000000000000a78
#
# JRE version: Java(TM) SE Runtime Environment (8.0_202-b08) (build 1.8.0_202-b08)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.202-b08 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [KinectPV2.dll+0x5549]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x0000000031bb5000):  JavaThread "main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1" [_thread_in_native, id=2680, stack(0x00000000456f0000,0x00000000457f0000)]

siginfo: ExceptionCode=0xc0000005, reading address 0x0000000000010001

Registers:
RAX=0x0000000000000001, RBX=0x0000000033145830, RCX=0x0000000000010001, RDX=0x00000000007e0000
RSP=0x00000000457ee2b0, RBP=0x0000000000000000, RSI=0x0000000000000005, RDI=0x0000000033145a68
R8 =0x00000000ffffffff, R9 =0x0000000000000001, R10=0x0000000000000000, R11=0x00000000457ee210
R12=0x0000000000000000, R13=0x0000000033eb74d8, R14=0x00000000457ee3a0, R15=0x0000000031bb5000
RIP=0x00007ffb34de5549, EFLAGS=0x0000000000010202

Top of Stack: (sp=0x00000000457ee2b0)
0x00000000457ee2b0:   0000000033145830 0000000033eb74d8
0x00000000457ee2c0:   0000000000000000 0000000028bcb8a0
0x00000000457ee2d0:   0000000028bcb8a0 00007ffb34de177a
0x00000000457ee2e0:   00000000457ee3a0 00000000457ee380
0x00000000457ee2f0:   0000000031bb51f8 0000000031bb5000
0x00000000457ee300:   00000000027983fd 00000000027b1df0
0x00000000457ee310:   0000000033eb74d8 0000000031bb5000
0x00000000457ee320:   0000000031bb5000 0000000031bb5000
0x00000000457ee330:   00000000457ee3f8 00000000027b17c2
0x00000000457ee340:   00000000457ee340 0000000033eb74d8
0x00000000457ee350:   00000000457ee3a0 0000000033eb9328
0x00000000457ee360:   0000000000000000 0000000033eb74d8
0x00000000457ee370:   0000000000000000 00000000457ee3a0
0x00000000457ee380:   00000000457ee3e8 0000000002797bd0
0x00000000457ee390:   0000000000000000 00000000027a7a40
0x00000000457ee3a0:   00000005ca9915d0 00000000457ee3a8 

Instructions: (pc=0x00007ffb34de5549)
0x00007ffb34de5529:   90 00 00 00 48 8b 4f 30 48 85 c9 74 0a 48 8b 01
0x00007ffb34de5539:   ff 50 10 48 89 6f 30 48 8b 4f 60 48 85 c9 74 0a
0x00007ffb34de5549:   48 8b 01 ff 50 10 48 89 6f 60 48 8b 8f 70 ff ff
0x00007ffb34de5559:   ff 48 85 c9 74 0d 48 8b 01 ff 50 10 48 89 af 70 

Register to memory mapping:

RAX=0x0000000000000001 is an unknown value
RBX=0x0000000033145830 is an unknown value
RCX=0x0000000000010001 is an unknown value
RDX=0x00000000007e0000 is an unknown value
RSP=0x00000000457ee2b0 is pointing into the stack for thread: 0x0000000031bb5000
RBP=0x0000000000000000 is an unknown value
RSI=0x0000000000000005 is an unknown value
RDI=0x0000000033145a68 is an unknown value
R8 =0x00000000ffffffff is an unknown value
R9 =0x0000000000000001 is an unknown value
R10=0x0000000000000000 is an unknown value
R11=0x00000000457ee210 is pointing into the stack for thread: 0x0000000031bb5000
R12=0x0000000000000000 is an unknown value
R13={method} {0x0000000033eb74e0} 'jniStopSignal' '()Z' in 'KinectPV2/Device'
R14=0x00000000457ee3a0 is pointing into the stack for thread: 0x0000000031bb5000
R15=0x0000000031bb5000 is a thread

Stack: [0x00000000456f0000,0x00000000457f0000],  sp=0x00000000457ee2b0,  free space=1016k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  KinectPV2.Device.jniStopSignal()Z+0
j  KinectPV2.Device.cleanDevice()Z+1
j  KinectPV2.KinectPV2.dispose()V+10
v  ~StubRoutines::call_stub
j  sun.reflect.NativeMethodAccessorImpl.invoke0(Ljava/lang/reflect/Method;Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+0
j  sun.reflect.NativeMethodAccessorImpl.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+100
j  sun.reflect.DelegatingMethodAccessorImpl.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+6
j  java.lang.reflect.Method.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+56
j  processing.core.PApplet$RegisteredMethods.handle([Ljava/lang/Object;)V+18
J 1966 C1 processing.core.PApplet.handleMethods(Ljava/lang/String;)V (36 bytes) @ 0x0000000002f82d6c [0x0000000002f829c0+0x3ac]
j  processing.core.PApplet.dispose()V+35
j  processing.opengl.PSurfaceJOGL$DrawListener.display(Lcom/jogamp/opengl/GLAutoDrawable;)V+206
j  jogamp.opengl.GLDrawableHelper.displayImpl(Lcom/jogamp/opengl/GLAutoDrawable;)V+62
j  jogamp.opengl.GLDrawableHelper.display(Lcom/jogamp/opengl/GLAutoDrawable;)V+2
j  jogamp.opengl.GLAutoDrawableBase$2.run()V+59
j  jogamp.opengl.GLDrawableHelper.invokeGLImpl(Lcom/jogamp/opengl/GLDrawable;Lcom/jogamp/opengl/GLContext;Ljava/lang/Runnable;Ljava/lang/Runnable;)V+203
j  jogamp.opengl.GLDrawableHelper.invokeGL(Lcom/jogamp/opengl/GLDrawable;Lcom/jogamp/opengl/GLContext;Ljava/lang/Runnable;Ljava/lang/Runnable;)V+72
j  com.jogamp.newt.opengl.GLWindow.display()V+90
j  com.jogamp.opengl.util.AWTAnimatorImpl.display(Ljava/util/ArrayList;ZZ)V+68
j  com.jogamp.opengl.util.AnimatorBase.display()V+16
j  com.jogamp.opengl.util.FPSAnimator$MainTask.run()V+216
j  java.util.TimerThread.mainLoop()V+221
j  java.util.TimerThread.run()V+1
v  ~StubRoutines::call_stub