ThomasLengeling/KinectPV2 #65: skeleton tracking

SimonBiggsUK, 2017-08-22:
I have the skeleton3d example working correctly in Windows 8. However, in Windows 10 instead of seeing the skeleton I just see the whole display window changing colour. Is there some issue with 3D skeleton tracking in Windows 10?

slowizzm, 2017-08-22:
I seem to remember the window being a solid color as well, but I had no problems with the library on windows 10. Here is kinectpv2 skeleton3d working with toxiclibs.

https://github.com/slowizzm/Processing-Gallery/tree/master/kinectParticles_spine

SimonBiggsUK, 2017-08-23:
Thanks
I've downloaded your example along with toxic and mesh libraries and installed everything but the app will not run as it can't find the class AttractionBehavior - which I think is part of the toxic library. Am I missing something?

slowizzm, 2017-08-29:
Yes, the AttractionBehavior class is part of toxi.physics2d.behaviors. Not sure why it would not be working properly. I do remember initially having issues with processing pointing to the library. You may just need to restart a few times until process locates it.

mbuccoli, 2017-09-27:
Hello, I had the same problem. It seems that the strokeWeight depends on the z-value of the joints, but it's not properly scaled (or maybe the function or the map scaling have changed across the version).
As a suggestion to test it the skeleton example, try to substitute every strokeWeight with a simple "strokeWeight(0.3)". Now I cannot see the bones, but the joints are just fine.

SimonBiggsUK, 2017-09-27:
I've found that turning off the setColor function to the kinematic object (PJBullet) attached to the joint incoming from Kinect the flashing problem has gone away. I am assuming I've not fixed the problem but have stopped the symptom. I can live with that for now.
//pJoint.setColor(color(Colour));

amozooz, 2017-10-27:
@SimonBiggsUK 
Hi I am having the same issue now. Can you elaborate more on where to comment out `pJoint.setColor(color(Colour));`? I couldn't find this line in this repository.

Many thanks!

SimonBiggsUK, 2017-10-27:
pJoint is only applicable if you are rendering the skeleton in the JBullet physics engine. Is that what you are doing? If not then it's not relevant, which is why you can't see the function call. What you can try is only drawing the bones and commenting out the joint drawing. This gives you a clean skeleton image. I found that after doing that I didn't have anymore artefacts. I think the problem is one of a mismatch in scaling, which affects the joint drawing but not the bone drawing. You might need to add some extra bones to incorporate all the joints. Set the strokeweight to something light, like 0.01. Here's the relevant code:

void drawBody(KJoint[] joints) {
  drawBone(joints, KinectPV2.JointType_Head, KinectPV2.JointType_Neck);
  drawBone(joints, KinectPV2.JointType_Neck, KinectPV2.JointType_SpineShoulder);
  drawBone(joints, KinectPV2.JointType_SpineShoulder, KinectPV2.JointType_SpineMid);
  drawBone(joints, KinectPV2.JointType_SpineMid, KinectPV2.JointType_SpineBase);
  drawBone(joints, KinectPV2.JointType_SpineShoulder, KinectPV2.JointType_ShoulderRight);
  drawBone(joints, KinectPV2.JointType_SpineShoulder, KinectPV2.JointType_ShoulderLeft);
  drawBone(joints, KinectPV2.JointType_SpineBase, KinectPV2.JointType_HipRight);
  drawBone(joints, KinectPV2.JointType_SpineBase, KinectPV2.JointType_HipLeft);
  // Right Arm    
  drawBone(joints, KinectPV2.JointType_ShoulderRight, KinectPV2.JointType_ElbowRight);
  drawBone(joints, KinectPV2.JointType_ElbowRight, KinectPV2.JointType_WristRight);
  drawBone(joints, KinectPV2.JointType_WristRight, KinectPV2.JointType_HandRight);
  drawBone(joints, KinectPV2.JointType_HandRight, KinectPV2.JointType_HandTipRight);
  drawBone(joints, KinectPV2.JointType_WristRight, KinectPV2.JointType_ThumbRight);
  // Left Arm
  drawBone(joints, KinectPV2.JointType_ShoulderLeft, KinectPV2.JointType_ElbowLeft);
  drawBone(joints, KinectPV2.JointType_ElbowLeft, KinectPV2.JointType_WristLeft);
  drawBone(joints, KinectPV2.JointType_WristLeft, KinectPV2.JointType_HandLeft);
  drawBone(joints, KinectPV2.JointType_HandLeft, KinectPV2.JointType_HandTipLeft);
  drawBone(joints, KinectPV2.JointType_WristLeft, KinectPV2.JointType_ThumbLeft);
  // Right Leg
  drawBone(joints, KinectPV2.JointType_HipRight, KinectPV2.JointType_KneeRight);
  drawBone(joints, KinectPV2.JointType_KneeRight, KinectPV2.JointType_AnkleRight);
  drawBone(joints, KinectPV2.JointType_AnkleRight, KinectPV2.JointType_FootRight);
  // Left Leg
  drawBone(joints, KinectPV2.JointType_HipLeft, KinectPV2.JointType_KneeLeft);
  drawBone(joints, KinectPV2.JointType_KneeLeft, KinectPV2.JointType_AnkleLeft);
  drawBone(joints, KinectPV2.JointType_AnkleLeft, KinectPV2.JointType_FootLeft);
}

void drawBone(KJoint[] joints, int jointType1, int jointType2) {
  strokeWeight(0.01f);
  line(joints[jointType1].getX(), joints[jointType1].getY(), joints[jointType1].getZ(),joints[jointType2].getX(), joints[jointType2].getY(), joints[jointType2].getZ());
}

amozooz, 2017-10-27:
Thanks for the quick reply!
I just tried your method, it works!

Thanks so much!!!

mauroibarra, 2019-12-21:
Thanks for the quick reply!
I just tried your method, it works!

Thanks so much!!!