ThomasLengeling/KinectPV2 #72: All I get is a black screen, console output is normal.

jo6fm06840616, 2019-08-19:
Hi, I'm doing research about Kinect V2.
I get some problem with that when I try to run the example from [KinectPV2](http://codigogenerativo.com/code/kinectpv2-k4w2-processing-library/) on Processing.

After I click "run" botton , All I get is:

console output :
64 windows 10
Loading KinectV2
Creating Kinect object ...
Setting Color Frame
Done Loading Kinect V2 
Version: 0.7.8

output:

[image]

[image]

Both example are from different example, All I get is a black screen (second one has frame data).
And my KinectV2 device has no light (I think the problem is because of it).

Please help me for solving this issue.

My environment are as follow:
Window 10 64bit
Kinect V2
Processing 3.4
KinectPV2 version : 0.7.8
SDK Browser v2.0 (Kinect for Windows)
USB 3.0

jo6fm06840616, 2019-08-20:
I resolved this problem already, thanks.

dani1294, 2019-11-18:
Hi, I'm having this same problem.
Could you let me know how you resolved it?

Thanks!

Suhurav, 2020-02-03:
How did you resolve it?

INKSec, 2023-05-18:
I am a little bit late, but i solved it.
I deleted the LibusbK driver i needed for openkinect for processing lib and switched back to the standard driver.
This one works with the official Kinect SDK and the KinectV2 library.